Antoxa1081/home-assistant-dess-monitor #1: Datalogger devcode 2376 integration

Antoxa1081, 2025-03-07:


Andyrotary, 2025-03-08:
сделал, оно?
[queryDeviceCtrlField.txt](https://github.com/user-attachments/files/19142537/queryDeviceCtrlField.txt)

Andyrotary, 2025-03-08:
> Responses
> 
> AC Line Mode [queryDeviceChartsFieldsEs.json](https://github.com/user-attachments/files/19133979/queryDeviceChartsFieldsEs.json) [queryDeviceParsEs.json](https://github.com/user-attachments/files/19133976/queryDeviceParsEs.json) [querySPDeviceLastData.json](https://github.com/user-attachments/files/19133977/querySPDeviceLastData.json) [webQueryDeviceEnergyFlowEs.json](https://github.com/user-attachments/files/19133978/webQueryDeviceEnergyFlowEs.json)
> 
> Battery Mode [queryDeviceParsEs.json](https://github.com/user-attachments/files/19133984/queryDeviceParsEs.json) [querySPDeviceLastData.json](https://github.com/user-attachments/files/19133985/querySPDeviceLastData.json) [webQueryDeviceEnergyFlowEs.json](https://github.com/user-attachments/files/19133986/webQueryDeviceEnergyFlowEs.json)

а что с этим делать? апдейт я установил

Antoxa1081, 2025-03-10:
Я выпустил новое обновление, в котором добавил поддержку всех возможных настроек с меню инвертора (приложения). Настройки стягиваются в поля постепенно, из-за ограниченности скорости запросов к апи. Так же добавил поддержку сенсоров энергии для батареи. Устранил проблемы с подсчётами в панели энергии.
Так же посмотрел виджет powerflow synsyk. В целом по минимуму можно все настроить, не хватает сенсоров потребления/выработки за последний текущий день. Но это вероятно проще будет реализовать через вспомогательные интеграции (хелперы).

Andyrotary, 2025-03-10:
Вау) установил, всего много добавилось, буду изучать) по виджету synsyk у меня получилось настроить примерно так. не хватает сенсоров по батарее, не заполняются проценты и по нагрузке мало данных. 

[image]

Andyrotary, 2025-03-10:
По панели энергии непонятно, вчера показывала корректно, сегодня вдруг какой то пик до 20 квт, хотя ничего такого не включалось... но может связано с обновлениями..

[image]

Andyrotary, 2025-03-10:
отлично, спасибо, исправил. теперь все корректно!

Antoxa1081, 2025-03-10:
По поводу напряжения батареи и процента заряда, все зависит от заявленных характеристик производителя батареи. Если производитель использовал "стандартное" напряжение для 12v / 24v / 48v, то для них применима такая схема распределения процентов от напряжения:

[image]

Например в случае использования батареи Kepworth 48v эти показатели совпадают, для них можно будет создать такой сенсор процента заряда.
В случае Dyness B4850 уже другие значения, там нижний порог 42-44v, а верхний 54v (для заряда 53.5v), а для Floating - 52.5v
То есть 100% заряженный значит >=52.5v. В целом можно сделать обычную интерполяцию от нижнего порога, до floating - 100%. Процент заряда который отдает SmartESS (DESS) считается для AGM свинца, он некорректный.

Antoxa1081, 2025-03-10:
По поводу сенсоров Daily Energy (для солнца, входа и выхода). Тут надо делать сенсор через helpers objects, чтобы он брал данные сенсора Energy (счетчика) за текущий день (от 00 начала суток до текущего момента). 
У меня получилось настроить без ежедневных метрик:

[image]
Силу тока на выходе с инвертора добавлю интеграции расчетом мощности с напряжением. 
А вот с батарей не очень понятно, вероятно нужен сенсор со знаками +/- для значений мощности и силы тока, чтобы виджет ориентировался в направлении движения энергии. Емкость батареи вручную надо задавать, с учетом отсечки по Cut Off (20% soc)

Есть у меня идея задавать емкость батареи на конфиг флоу (когда учетные данные аккаунта задаем), для каждого инвертора указывать емкость установленных батарей. Имея эту информацию можно будет высчитывать сколько закачало/выкачало в акум (сбрасывать цикл закачки после полного заряда, определять его по напряжению). Тогда получится точно отслеживать в Wh текущую емкость аккумулятора, и вычислять настоящий процент по емкости, а не по напряжению (так как напряжение проседает при нагрузке батареи, и создает "ложные" показания)

Antoxa1081, 2025-03-11:
В настройках карточки sunsyk
```yaml
battery:  
  invert_power: false
  invert_flow: true
entities:
    battery_power_190: sensor.battery_power_state
```

/config/helpers

[image]

Шаблон сенсора

Название Battery Power State
```python
{% set battery_charge_power = float(states('sensor.INVERTERNAME_battery_charge_power')) | float(0) %}
{% set battery_discharge_power = float(states('sensor.INVERTERNAME_battery_discharge_power')) | float(0) %}
{% if (battery_discharge_power) > 0 %} {{ battery_discharge_power * -1 }}
{% elif (battery_charge_power) > 0 %}
{{ battery_charge_power }}
{% else %}
{{ 0 }}
{% endif %}
```

[image]

Andyrotary, 2025-03-12:
sensor.INVERTERNAME_battery_charge_power тут я должен заменить invertername на данные своего логгера?

Andyrotary, 2025-03-12:
Все работает! Спасибо большое
[image]

Antoxa1081, 2025-03-12:
Вероятно сегодня выпущу новое обновление, в котором отключу управление "настройками" инвертора, оставлю только унифицированные 2 опции (режим работы, режим зарядки, возможно ещё какую-то одну настройку). Остальные придется убрать, так как у меня дата логгер на одном из 3х устройств не выдерживает такой "нагрузки" опроса параметров раз в 10 минут, так как он в лайве ходит к инвертору спрашивать настройки, а когда у дата логгера их много (настроек, у моего около 20 доступно), даже поочередное получение вызывает у него какие-то странные сбои. У меня на днях начал даталоггер пропадать из сети, а сегодня он у меня вовсе перестал слать обновления на облако (при этом в сети онлайн), может это совпадение у меня какое-то с контактами кабеля, или сбои в облаке.. смогу только на выходных точно узнать. Пока для безопасности думаю отключить эту функцию.

Andyrotary, 2025-03-12:
Да, я заметил, что начало сильнее подвисать, особенно на сайте. кстати, заметил что у дессмонитора есть платные функции, которые заметно учащают отправку данных . стоит вроде не дорого, но не понятно на какой период работает оплата. у них как то расплывчато написано, четких сроков не нашел

Antoxa1081, 2025-03-12:
У них неправильный перевод на английский, если переключиться на китайский и перевести переводчиком, там месячная оплата от 4$ до 8$ (RMB/USD). Обещают что будет чаще отправлять на облако, вместо 1 раза в 5 минут, от 1 раза в 30 сек до 1 в 5 секунд. Слабо верится что это будет корректно работать) У них неправильно считается половина статистики на сайте и в приложении, а с тем учетом что я увидел в ответах апи.. :)

Andyrotary, 2025-03-12:
пусть гуляют) в таком варианте уже лучше солар ассистан взять за 60 дол единоразово и получать нормальные показания и  с нормальной частотой

Antoxa1081, 2025-03-12:
О, у меня разлагал даталоггер, после удаленной перезагрузки роутеров, возможно у меня была проблема с провайдером интернета с доступом к серверу облачному. То что тормозит в браузере веб версия DESS это нормально, если у вас запущен хоум ассистант интеграция и вы пользуетесь с одного айпи адреса провайдера интернета, сервера ESS режут скорость запросов по вашему айпи. Если зайти с мобильного интернета отдельно, будет быстро отвечать.

Antoxa1081, 2025-03-19:
> пусть гуляют) в таком варианте уже лучше солар ассистан взять за 60 дол единоразово и получать нормальные показания и с нормальной частотой

@Andyrotary 
сегодня запустил на pi4B Solar Assistant и подключил к EaSun SMH 3

[image]

Подключал консольным кабелем по типу 

[image]
но в его изначальном виде работать не будет, RJ45 коннектор надо переобжать по пинам соответственно вашему инвертору

Для моего инвертора распиновка RJ45 такая:
[image]

Распиновка для FTDI USB RS232 RJ45:

[image]

Итоговое подключение:

[image]

На моем инверторе используется пины 1 2 8 RJ45
Которые соответствуют 3 6 5 на FTDI USB RJ45

[image]
в Home Assistant через MQTT

[image]

 
[image]

Andyrotary, 2025-03-19:
круто) у меня Anenji, это клон еасуна, надо будет попробовать. и как работает? лучше чем со даталоггером?

Antoxa1081, 2025-03-19:
В целом намного лучше, обновления данных каждые 2-3 секунды, встроенная инфографика и метрика заряда акума с симуляцией BMS. А самое главное не зависит от ненадежного китайского даталоггер сервера.

closed 2025-03-29 by Antoxa1081