Antoxa1081/home-assistant-dess-monitor #3: Почему не выдает данные PV Power?

wsxshop24, 2025-03-29:
Установил ваш репозиторий, мне очень понравился. Прост в установке и использовании.
Есть только вопрос, почему не выдает данные PV Power? На вашем скриншоте также видно, что PV Power 0W
Как это исправить?
Это ваш скриншот
[image]

А это то что у меня
[image]

Antoxa1081, 2025-03-29:
Я делал скриншоты поздно ночью, когда солнца не было) PV Power у меня отображается нормально, скорее всего у вас ревизия даталоггера имеет другое название поля для pv power, https://github.com/Antoxa1081/smart-ess-api-gateway/issues/5#issuecomment-2704532980 

По этой инструкции можете мне прислать мне ответы запросов, сможем посмотреть в каком поле это значение.

wsxshop24, 2025-03-29:
Я буду вам очень благодарен за помощь.

[image]

[image]

Antoxa1081, 2025-03-29:
Сможете пожалуйста в таком формате передать данные? https://github.com/Antoxa1081/smart-ess-api-gateway/issues/5#issuecomment-2704567990

В виде файлов в которых ответ от запроса записан текстом..
В ваших скриншотах я пока не увидел PV Power, или что-то похожее на него.
Найдите запрос webQueryDeviceEnergyFlowEs, возможно там есть эта информация

wsxshop24, 2025-03-29:
[Response.txt](https://github.com/user-attachments/files/19516669/Response.txt)

[Payload.txt](https://github.com/user-attachments/files/19516670/Payload.txt)

wsxshop24, 2025-03-29:
Пожалуйста, сообщите когда будут какие-то результаты. Спасибо.

Antoxa1081, 2025-03-29:
@wsxshop24 установите обновление [0.1.2](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.2)

wsxshop24, 2025-03-29:
Спасибо большое! Вы настоящий гений своего дела 🥇 Все получилось!
Ещё бы вопрос решить с батареями, вообще цены вам не было.

[image]

Antoxa1081, 2025-03-29:
Discharge current и Discharge  power получится добавить, а вот Charge Power скорее всего не получится, так как у вас даталоггер не выдает данные о напряжении зарядки в запросе SPLastData

wsxshop24, 2025-03-29:
Добавьте, пожалуйста, хоть что получится.

Antoxa1081, 2025-03-29:
@wsxshop24 добавил в обновлении https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.3

wsxshop24, 2025-03-29:
Спасибо вам огромное!!! Все работает.

closed 2025-03-29 by Antoxa1081