Antoxa1081/home-assistant-dess-monitor #4: Set up new device

robertio, 2025-03-29:
Hi 

thanks for Dessmonitor HomeAssistant integration.
i have a Easun ISolar SMW 11Kw hybrid inverter, and would like to see inverter data in HA.
Desmonitor login success. Protocol: 2451 Device address: 1
DESS Monitor latest 0.1.3  installed and Integration entries is set up. 
Login success. 
But no device set up.
No devices or entities

[image]

Can you pls help how to debug, what to do? 

Best Regards,
Robert

Antoxa1081, 2025-03-29:
Hi, I'm seeing problems on ESS servers today, not everything is loading for me either (not only in integration, but also in the browser via https://www.dessmonitor.com/). Try turning on some vpn/proxy, everything works fine for me through cloudflare warp. This problem periodically occurs when working with smart ess. It is the main reason for the slow display of data in the mobile application. Sometimes it happens that the data logger cannot send data to their servers either. Perhaps the problem is also with DNS.

Antoxa1081, 2025-03-29:
If possible, show your device with a screenshot from the https://www.dessmonitor.com , perhaps it has a different device class (as an energy storage device, not as an inverter).

robertio, 2025-03-29:
Hi Antoxa, 
Thanks for quick response.
Which screen do you need? 

[image]
[image]

Antoxa1081, 2025-03-29:
I see your device is in offline status (gray circle)

[image]

Status must be online (green)
[image]

This integration load only online devices, because in offline status api return data only in historical view (not current last data)
Can you confirm what your device wifi data logger works correctly?

robertio, 2025-03-29:
>Status must be online (green)
ahhh. i didnt know that. thanks for clarify. this is key information.

Im fully confused with this Dessmonitor. 
Sometimes it is online and have flow, but sometimes offline. 

So i deleted the device, and adding back take hours. 
Dessmonitor is horrible software. :)

**Now i see it Normal "state".** 

[image]

And now i have just successfully Reload Integration entries.   :)    
Many thanks to help!

[image]

there are Inverter Values. 

[image]

So your

Antoxa1081, 2025-03-29:
Yes, they haven't done it very well there, each data logger has its own field names that mean the same thing, sometimes just a letter is missing, or the case is different... it's impossible to adequately reduce all data loggers to fields with uniform names (at least I'm trying).

closed 2025-03-30 by Antoxa1081