Antoxa1081/home-assistant-dess-monitor #5: Пропал объект "Anenji 6200 Output Priority" после обновления.

reallist93, 2025-03-30:
Добрый день.
Инвертор Anenji 6200, протокол 2376.
После обновления на версию 0.1.3, в HA пропал объект "Anenji 6200 Output Priority" ( select.anenji_6200_output_priority ) .
Как его вернуть?
P.S. Ранее этот пункт имел всего 3 значения на выбор: "Utility, Solar, SBU", но не было пункта  "SUB". Можно его добавить настройки ?
Заранее спасибо.

Antoxa1081, 2025-03-30:
Удалите запись настройки, и добавьте заново. Вероятно внутренние настройки интеграции несовместимые стали. При добавлении поставьте галочку (под полями логина)

reallist93, 2025-03-30:
Спасибо, помогло.. )
Почему нет пункта "SUB" , который присутствует в настройках инвертора ?
Можно его добавить?

reallist93, 2025-03-30:
Нашел , в настройках, в пункте :  "Anenji 6200 SET Output priority" возможность выбора режима "SUB", этого достаточно..
Ещё раз спасибо.
P.S. Его нет в HA в "Управление".

[image]

Antoxa1081, 2025-03-30:
А, понятно, там может не быть такого, из-за неполной адаптации под ваш девкод. Настройки универсальные, должны тянуться все возможные как и в приложении.

closed 2025-03-30 by reallist93