Antoxa1081/home-assistant-dess-monitor #6: Unable to read 'charge_priority'

peejaydub, 2025-04-03:
I have a cheap PowMr Inverter.  I am unable to set the charge mode to 'Solar Only' via the Dessmonitor site (or ESSMonitor App), I can only set it via the inverter menus.  If I set it manually, on the SmartESS site the status reports as '--'.
PowMr Support have been unable to resolve this but FYI the original Inverter was DoA so they sent me a new controller board, and the Wifi PlugPro 05 I got from another inverter vendor so I presume one of these is the issue.  I was hoping that an HA integration might find a workaround but instead the plugin fails to initialise properly if I have 'Solar Only (OSO)' set.  I get the following error

[dess_monitor-01JQY2JBERCRS3SM8DJAVTXHZN-Q004416474758709AD05-656b3eb330da7122ce36e2b9f21fe4b7.json](https://github.com/user-attachments/files/19591014/dess_monitor-01JQY2JBERCRS3SM8DJAVTXHZN-Q004416474758709AD05-656b3eb330da7122ce36e2b9f21fe4b7.json)

:

```
2025-04-03 16:55:22.157 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/dess_monitor/sensor.py", line 783, in _handle_coordinator_update
    self._attr_native_value = resolve_charge_priority(data, device_data)
                              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/api/helpers.py", line 182, in resolve_charge_priority
    mapper[
    ~~~~~~^
        next(
        ^^^^^
            (x for x in data['last_data']['pars']['bt_']
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
             if
             ^^
             x['id'] == 'bt_charger_source_priority'), {'val': None})['val']]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'N/A'
```

I'm hoping you may be able to see if you can work around the unexpected result here and allow initialisation to complete, it would be useful to be able to set/overwrite the value still even if it can not be read and maybe add a bit more info to the debug logs to confirm what value is really coming back (not sure if it's actually '--' or if the site is translating that).

Antoxa1081, 2025-04-05:
Unfortunately, your data logger does not fully support your equipment, so the API receives this response 
```json
{
  "id": "bt_charger_source_priority",
  "par": "Charger Source Priority",
  "val": "N/A"
}
```
This happens because the data logger has different versions of protocols for working with different vendors that produce inverters.
[image]

Try to find a data logger with a different firmware version, or from a different seller. For example, I have 4 data loggers, and 2 of them have different firmware versions, due to which the data loggers provide me with different sets of available settings for the same inverter.

In the upcoming update I will make a fix to launch the hub with your device, but I will not be able to help with changing the battery charging mode through integration.

peejaydub, 2025-04-05:
Thanks, that's helpful.  I don't mind buying another module if I know it works, but the vendor of the last one assured me this one was compatible :). Can you help me understand your screenshot though, it looks a bit like there are 3 main types of firmware out there - Voltronic, SRNE & Sumry and I need the Voltronic one, is that your understanding too?

Antoxa1081, 2025-04-05:
Yes, that's right. Your inverter may be from a different vendor, but it shouldn't affect the data logger protocol much. This list is approximate and may not include other end inverter distributors. More info about vendors you can check on https://solar-assistant.io/help/easun/configuration (source of screenshot table) or on solar community forums

Antoxa1081, 2025-04-05:
Resolved in [v0.1.7](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.7)

closed 2025-04-14 by Antoxa1081