Antoxa1081/home-assistant-dess-monitor #7: Feature request: add battery capacity in percents

LehValensa, 2025-04-05:
Hi! Could you **add battery capacity** in percents to the list of sensors?
See below for the actual response to webQueryDeviceEnergyFlowEs from my inverter (Powmr 6.2KW):

<code>{"err":0,"desc":"ERR_NONE","dat":{"brand":0,"status":0,"date":"2025-04-05 06:30:22","bt_status":[{"par":"bt_battery_capacity","val":"21.0000","unit":"%","status":2},{"par":"battery_active_power","val":"-0.0360","unit":"kW","status":0}],"pv_status":[{"par":"pv_output_power","val":"0.0010","unit":"kW","status":0}],"gd_status":[{"par":"grid_active_power","val":"0.6940","unit":"kW","status":1}],"bc_status":[{"par":"load_active_power","val":"0.6540","unit":"kW","status":-1}],"ol_status":[{"par":"oil_output_power","val":"0","status":0}],"we_status":[{"par":"wind_output_power","val":"0","status":0}],"mi_status":[],"mt_status":[],"wp_status":[]}}
</code>

It looks like the value of bt_battery_capacity can be parsed the same way you do with *power sensors in the given query.
PS: I can assist you with testing, if needed.

regards
Alex.

Antoxa1081, 2025-04-05:
Hello, in the future update it is planned to add "raw" sensors, which will allow to transfer to home assistant all fields, which the data logger gives without standardization. It's currently available in a limited preview mode on [v0.1.6 pre release](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.6).
Now if you have the opportunity, get the diagnostic file of the added device in the integration
[image]
This will allow me to get more information on other requests to understand where this parameter can be obtained additionally.

Antoxa1081, 2025-04-05:
Added in [v0.1.7](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.7)

jello1974, 2025-04-05:
Sadly after update I can't make it work aggain. Tried do uninstall and reinstall 0.1.6 but still getting error:
Unexpected error fetching My sensor data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/coordinator.py", line 127, in _async_update_data
    query_device_output_priority_results = await asyncio.gather(*query_device_output_priority)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/api/helpers.py", line 345, in get_inverter_output_priority
    return map_param_value[result['val']]
           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'Utility'

jello1974, 2025-04-05:
After some debugging I changed in helpers.py 
async def get_inverter_output_priority(token: str, secret: str, device_data):
    match device_data['devcode']:
        case 2341:
            map_param_value = {
                'Utility first': 'Utility',
                'Utility': 'Utility',
                'Solar first': 'Solar',
                'Solar': 'Solar',
                'SBU': 'SBU',
                'SBU first': 'SBU',
                None: None
            }
            # param_value = map_param_value[value]

            param_id = 'los_output_source_priority'
        case 2428:
            map_param_value = {
                'Utility first': 'Utility',
                'Solar first': 'Solar',
                'SBU': 'SBU',
                'SBU first': 'SBU',
                None: None
            }
            # param_value = map_param_value[value]

            param_id = 'bse_output_source_priority'

        case _:
            return
    result = await get_device_ctrl_value(token, secret, device_data, param_id)

    return map_param_value[result['val']]

and now is working aggain

Antoxa1081, 2025-04-05:
@jello1974 it's very strange, since in the latest updates I didn't change anything related to this sensor in which you now have an error. Perhaps you didn't get this type of fields before (for example, it was in SBU mode). Now resolved in [v0.1.8](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.8)

LehValensa, 2025-04-05:
> Hello, in the future update it is planned to add "raw" sensors, which will allow to transfer to home assistant all fields, which the data logger gives without standardization. It's currently available in a limited preview mode on [v0.1.6 pre release](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.1.6). Now if you have the opportunity, get the diagnostic file of the added device in the integration <img alt="" width="40%" src="https://private-user-images.githubusercontent.com/30043921/430611936-e8ab0f7c-2165-420e-a3d7-83d3dac273fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDM4ODcwMjEsIm5iZiI6MTc0Mzg4NjcyMSwicGF0aCI6Ii8zMDA0MzkyMS80MzA2MTE5MzYtZThhYjBmN2MtMjE2NS00MjBlLWEzZDctODNkM2RhYzI3M2ZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA1VDIwNTg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM3YThiZTBlZDEwMjQ4ODUwMGNiNGM5MGEzN2Q0NDE3MDY0OTcwMDAyYTkwMDZkMzNiYjNlYjQ0ZjM0NWE1MDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SZSPeIoIBTvRbwGvySghdinoaru7I6PVs7uqqb9bdww"> This will allow me to get more information on other requests to understand where this parameter can be obtained additionally.
> 
> But keep in mind, this sensor is provided by the data logger based on the inverter calculation, this may not be accurate, since by default the inverter calculates by interpolation of voltage for a lead battery. But if in your case a lithium battery with BMS feedback via the RS232/RS485 protocol is used, this may be normal.

Thanks a lot for the prompt update! It works like a charm. Now my Powerflow card is complete.
The remaining battery percentage is shown correctly. And yes, LiFePo battery with BMS is connected to the inverter by cable via RS485, so I believe it provides the correct value of battery capacity.

[image]
I used https://github.com/slipx06/sunsynk-power-flow-card for visualization.

See the attached diagnostic file you've requested.
[dess_monitor-01JQZ6H0MXQ1AWJ81MJ58B6A5N-Q0044116603229094801-0f68cc52ff8ead012f57519522adece9.json.gz](https://github.com/user-attachments/files/19618651/dess_monitor-01JQZ6H0MXQ1AWJ81MJ58B6A5N-Q0044116603229094801-0f68cc52ff8ead012f57519522adece9.json.gz)
Do not hesitate to ask me to do more tests.

jello1974, 2025-04-05:
Now that you mention it I did observed in previous versions that the sensor only updates when the inverter was in SBU mode, I did not get any error but sensor does not update if inverter was in any other mode than SBU.