Antoxa1081/home-assistant-dess-monitor #8: Configuration appear not read

TAFESH10, 2025-04-06:
I do all things and my inverter was added I can configure the settings but I cant read the data I didn't see like power and battery percentage 

[image]

Antoxa1081, 2025-04-06:
Please provide your device diagnostic data (download file) and if possible debug logs while integration initialize

TAFESH10, 2025-04-06:
[dess_monitor-01JR66JAN9342VACDXX9XZ1EHY-Home-314f0120bde39a212c4751cd194e642b.json](https://github.com/user-attachments/files/19622927/dess_monitor-01JR66JAN9342VACDXX9XZ1EHY-Home-314f0120bde39a212c4751cd194e642b.json)

TAFESH10, 2025-04-06:
you mean when i add it i remember that I was see 2 checkbox I check on the two

Antoxa1081, 2025-04-06:
Try to use without check on "raw sensors"

TAFESH10, 2025-04-06:
now it works thank you 
i want to ask I can use automation in it ?

Antoxa1081, 2025-04-06:
Yes, you can with home assistant capabilities

TAFESH10, 2025-04-06:
thank you so much

closed 2025-04-06 by TAFESH10

Antoxa1081, 2025-04-06:
Thanks for providing diagnostic data, I will do a detailed analysis to make corrections for raw sensora mode in a future version.

TAFESH10, 2025-04-06:
Welcome