Antoxa1081/home-assistant-dess-monitor #9: Protocol 2546

toyota4wd, 2025-04-07:
Добрый день. 
Установил интеграцию, все очень круто сделано. Снимаю шляпу за проделанную работу. 
Заметил несколько нюансов. В версии 15 протокол 2546 отлично подтягивался, правда некоторые сенсоры не работали. 
В 18 версии после инсталляции уже не находит сенсоры. 
Решил проблему установкой сначала 15 версии а потом обновлением до 19.. Тогда хоть как-то сенсоры показывает. 

Вопрос. Как можно добавить протокол 2546 ?

Что для этого нужно, какие данные нужны ? 

Спасибо. 

П.С. Хотелось бы полноценно пользоваться интеграцией.

Antoxa1081, 2025-04-07:
Добрый день, попробуйте удалить конфигурацию и заново добавить её, возможно произошел конфликт миграции старый настроек на новый вид. Чтобы полноценно внести ваши протокол, на странице устройства есть кнопка "скачать диагностический файл", с его помощью я смогу адаптировать сенсоры под ваш девкод. Ещё вы можете попробовать при добавлении конфигурации (или настройке) включить "raw sensors", это создаст сенсоры напрямую с именами от даталоггера, с префиксом "Raw".

toyota4wd, 2025-04-07:
> Добрый день, попробуйте удалить конфигурацию и заново добавить её, возможно произошел конфликт миграции старый настроек на новый вид. Чтобы полноценно внести ваши протокол, на странице устройства [...]

К сожалению не могу найти "скачать диагностический файл" ))).  Пните плиз по направлению где эта "кнопочка" )) 
В каком он формате? 
Спасибо

Antoxa1081, 2025-04-07:
https://github.com/Antoxa1081/home-assistant-dess-monitor/issues/7#issuecomment-2780649484

Можете посмотреть тут на скриншоте показано

toyota4wd, 2025-04-07:
Спасибо. )) 

[dess_monitor.json](https://github.com/user-attachments/files/19630528/dess_monitor.json)

Antoxa1081, 2025-04-09:
В ближайшее время проверю совместимость данных и в ближайших обновлениях будут исправления. Можете показать скриншотами какие сенсоры и как отображаются сейчас у вас?

Antoxa1081, 2025-04-14:
@toyota4wd 
Попробуйте использовать эту пре-релизную версию [v0.2.0](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.2.0)

[image]
[image]

toyota4wd, 2025-05-13:
Спасибо большое за ответы. 
Нашел такую интересную штуку в дессмониторе, что можно отправлять запрос прямо на инвертор вкладка "Debug", и ответ приходит довольно быстро и с актуальными данными (обновление не раз в 5 минут).  
Есть ли возможность как-то отправлять через апи запрос и получать ответ от ивертора? 
Если что у меня есть мануал от двух инверторов по запросам/ответам от производителя.

Antoxa1081, 2025-05-13:
Да, можно попробовать. Какие модели инверторов у вас? Можете мне отправить инфо на электронную почту antoxa1081@gmail.com, или же в телеграм группе https://t.me/+cd7t95GtiRU4MTgy