Antoxa1081/home-assistant-dess-monitor #10: Not a issue but it would be great if its possible

philspain72, 2025-04-09:
Thanks for your great work its really good. 
If it would be possible to use something like the elfin ew10a to use the same data locally, that would be amazing.
Thanks again for the great work

Antoxa1081, 2025-04-09:
Thank you for your feedback, I'm glad you found it useful.
There is such a project (but it is paid) which provides a solution for local monitoring via modbus - https://solar-assistant.io/
Each inverter has its own vendor (there are about 4-5 of them). And each has different data reading parameters, so you will need to adapt it to your own model yourself. 
Open source projects for example: 
https://github.com/odya/esphome-powmr-hybrid-inverter 
https://github.com/suletom/EASUN-ISOLAR-SMX-II-CONTROL

philspain72, 2025-04-09:
Thanks, I have looked at solar-assistant in the past, but the problem for me is having to have another raspberry-pi. I have tried using the esphome project but I have spent loads on them but I have never actually got any of them working on Home Assistant. I have quite a few Epever charge controllers and they work really well on the Elfin ew11a and really easy to setup. 
I know that what I was  asking was out of your original goal. I was not expecting anything just saying that it would be nice. 
Thanks Agan

Antoxa1081, 2025-04-09:
https://github.com/suletom/EASUN-ISOLAR-SMX-II-CONTROL
This repository has an interesting solution for extracting data via a modbus bridge datalogger. I tried it, I couldn't set it up correctly. It works by connecting via a local network (or hotspot network), if you try to use the Energy Mate android application without logging into an account, there is a settings button (connect to the data logger's access point) and inside the settings window there is "information" that is requested directly from the data logger (without the cloud). Perhaps this is one of the options for solving the problem, using existing WiFi Plug Pro equipment (since there is actually a modbus TCP converter there)

philspain72, 2025-04-09:
Yes i will have a look it's managing to get the data into Home Assistant. The elfin ew10a works the same but they seem a lot easier to get the data from it into HA at least for the epever.
All I need an easy way to connect my pylontech us2000c into Home Assistant without using the esphome. I have tried at least 4 different times but never had any luck with any of them.

closed 2025-04-17 by Antoxa1081

philspain72, 2025-04-17:
I will do that at the moment. I have 6 Epever charge Controllers, and they are really easy with the Elfin ew11a and work really well. I have decided to buy the Solar-Assistant to see if it works like they claim. If it does, I decided that might be the best option for me because it workout cheaper in the long run. I have 2 EASun Inverters and Pylontech batteries which to me at least seem too hard to get running without it. I have tried  to connect the batteries with loads of differemt ways on the internet no it seems no one actually shows you in say like a youtube video how to do it even there is one that uses the elfin ew 10a but it only show you part of it not step by step guide.

Antoxa1081, 2025-05-22:
@philspain72 Hi, I just came to the idea of ​​integration with Elfin (or any other TCP WiFi Modus Bridge), in the latest updates (v0.2.5+) I have already started doing direct polling via the cloud datalogger inverter commands like QPIGS but via the native inverter protocol. Now I want to add an adapter to connect to Elfin via a local network. Will this be relevant for you now? I will need your help to debug some requests via Elfin, since at the moment I do not have such equipment yet.

philspain72, 2025-05-22:
> [@philspain72](https://github.com/philspain72) Hi, I just came to the idea of ​​integration with Elfin (or any other TCP WiFi Modus Bridge), in the latest updates (v0.2.5+) I have already started [...]

I decided to buy Solar Assistant. But I will help you testing or any other help you need

philspain72, 2025-05-22:
Yes that would be a great solution. Like I said, I'm up for helping in any way I can, testing on my equipment.

Antoxa1081, 2025-05-22:
@philspain72 

1. Configure EW10 or 11 for RS-232 Transparent Mode

In the web interface:

Serial Interface: RS-232

Baud Rate: Match inverter (typically 2400, 9600, or 115200)

Working Mode: TCP Server

Port: e.g., 8899

Serial Protocol: None (for transparent mode)

---

2. Python Script to Send QPIGS and Read Response

```python
import socket
import time

# Replace with your actual EW11 IP and port
HOST = '192.168.0.18'
PORT = 8899

# QPIGS command — ends with carriage return
command = b'QPIGS\x0D'  # or b'QPIGS\r'

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    s.settimeout(3)
    s.connect((HOST, PORT))
    
    s.sendall(command)
    time.sleep(0.1)  # slight delay may help
    response = s.recv(512)

    print("Raw response:", response)
    try:
        print("Decoded:", response.decode('utf-8').strip())
    except UnicodeDecodeError:
        print("Non-UTF8 response, hex:", response.hex())
```

---

3. What the Output Looks Like

Typical QPIGS output:

(230.0 50.0 230.0 50.0 0012 0050 009 435 13.0 000 100 47.0 0000 0000 010 1 0 01101000 110 )

Each value corresponds to a different inverter stat — voltage, frequency, power, etc.