Antoxa1081/home-assistant-dess-monitor #11: Devcode 6416 with issues

pdrobek, 2025-04-14:
Hi i have issue

[image]

device: 6416 

how can i fix it?

Antoxa1081, 2025-04-14:
Can you provide your debug log from home assistant?
And provide diagnostics file for entry 

[image]

Antoxa1081, 2025-04-14:
Try to use prerelease version [v0.2.0](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.2.0)

[image]
[image]

pdrobek, 2025-04-14:
on 0.2.0 version i have got only

[image]

Antoxa1081, 2025-04-17:
@pdrobek can you please provide debug logs to analyze the problem? Maybe it is a problem with connecting to the ess cloud

pdrobek, 2025-04-18:
Yes it was problem with CORS-header missing in request, after sort this out the next issue was overloaded ess site. After many,many tries and switch-on debug mode in your hacs addon I finaly got my hybrid invertets back online.

closed 2025-04-18 by pdrobek