Antoxa1081/home-assistant-dess-monitor #14: Missing SUB Mode

brunoclc, 2025-04-22:
[image]

There is no SUB Mode on Anenji using Protocol 2376

brunoclc, 2025-04-22:
Already enabled, and I can setup all modes avaiables, but SUB isnt there...

Antoxa1081, 2025-04-22:
This option must be available in SET * controls in diagnostic/settings sections. In control sectition SUB = SOLAR

brunoclc, 2025-04-22:
No, my inverter has 4 options: UTI, SOL, SUB, SBU as you can see on my SmartESS

[image]

brunoclc, 2025-04-22:
Under settings it is avaiable... So, its just a bug in control options

 [image]

Antoxa1081, 2025-04-24:
This is not a bug, since there are 2 different ways to control implemented here. 1 - standardized, works only for the devcode settings known to me, and 2 - general, which duplicates the logic of the settings from the official website of Dess

closed 2025-04-24 by Antoxa1081