Antoxa1081/home-assistant-dess-monitor #17: protocol 2499

DennisAppelman, 2025-05-10:
Thanks a lot for this integration.
most of the sensors do show values. 
some of them only as RAW sensor. like the load and grid voltage.
the inverter uses protocol 2499.

is it possible to show the “ Inverter Load” and “Grid In Voltage” values also instead of the RAW values?

“Inverter INV Temperature” and “Inverter DC Temperature” are also “unknown”.

last question: what does the configuration option “dynamic settings” do?

thanks a lot and regards
Dennis

Antoxa1081, 2025-05-13:
Thank you for your feedback, it's good that you found it useful.

Dynamic settings - includes the ability to control the inverter settings, broadcasting all available parameters to the home assistant integration interface. With their help, you can change the operating mode, charging speed, etc., which was available in the mobile application or the website of the Dess Monitor. (For example, you can create an automation schedule for changing the charging mode).

Temperature sensors provide information only if it can be read from the Dess data API. This integration can only transmit data that is available to Dess, nothing more. The set of data and capabilities depends on different inverter firmware, from the data loggers themselves and their firmware versions, including.

Raw sensors were made to cover the differences in parameter names from version to version of inverters/firmware/data loggers. Unfortunately, each manufacturer in the data logger specifies the parameter names in its own way, which is why it is quite difficult to reduce them to some general form if you have not seen this data in advance.

If you can provide diagnostic data that can be obtained on the page of your device in the home assistant integration (download diagnostic data button under the info version, serial number block), then in the next update I will be able to combine the sensors of your data logger/device into common fields (not raw).

DennisAppelman, 2025-05-17:
It would be really nice if you could add protocol 2499 to the  next update.
i added the diagnostics to this message.

Thanks in advance...

[dess_monitor-01JVE4M93F75MNBEJ8YZ0TKG7Q-TommaTech-b2943115901fcef571879587006c32f5.json](https://github.com/user-attachments/files/20262834/dess_monitor-01JVE4M93F75MNBEJ8YZ0TKG7Q-TommaTech-b2943115901fcef571879587006c32f5.json)

Antoxa1081, 2025-05-19:
thank you for your feedback, in the next update I will try to adapt it to your devcode. also now you can try new functions in version 0.2.4 which was released today. such as direct data update via QPIGS protocol

DennisAppelman, 2025-05-20:
thanks a lot..
does the direct data update via QPIGS protocol connect directly to the inverter or is it via the cloud?
I tried the direct data update, it worked but after a while i could not connect anymore to dessmonitor.
i could connect only from a different ip addess.  could this be connected to the direct data update?

Antoxa1081, 2025-05-20:
HA integration -> DESS API -> Wifi Datalogger -> Inverter

Direct data update send command to inverter via DESS API cloud, which delivery message to inverter via datalogger connection. Without using direct update, ha integration requests the latest data from the cloud dess API, and the cloud receives it every 5 minutes from the data logger (scheduled updates).

Direct update allows to increase the frequency of data update to an interval of 10 seconds, in contrast to the 5-minute interval of the cloud. Such frequent data updates cause "restrictions" on the IP address from which the request to the Dess cloud is sent. (These restrictions are imposed by the Dess cloud)

closed 2025-07-01 by Antoxa1081