Antoxa1081/home-assistant-dess-monitor #18: protocollo 2462

saxo1044, 2025-05-17:
ciao saresti cosi gentile da fare un protocollo per il mio inverter che usa il 2462. grazie

Antoxa1081, 2025-05-17:
[image]

[image]

Using HACS repo settings install another version - [v0.2.1](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.2.1) 

When configure enable toggle - raw sensors.
You will be able to see sensors with prefix Raw*

saxo1044, 2025-05-17:
Non riesco a cancellare la 0.2.2
Come si fa'?

Antoxa1081, 2025-05-17:
Just select this version and download. Restart HA. Readd config entry for your account

saxo1044, 2025-05-17:
> Seleziona questa versione e scaricala. Riavvia HA. ​​Aggiungi la voce di configurazione Readd per il tuo account.

Non riesco ad arrivare a questa schermata 

[image]

Antoxa1081, 2025-05-17:
url to hacs repo

http://your_ip_address:8123/hacs/repository/943574315

[image]

next

[image]

next

[image]

saxo1044, 2025-05-18:
Scusami sono completamente un neofita,non riesco a cancellare la 0.2.2 e istallare la 0.2.1 mi puoi seguire passo passo? Grazie infinitamente.

Antoxa1081, 2025-05-19:
Now you can install latest version (0.2.4) released today and try to setup integration again