Antoxa1081/home-assistant-dess-monitor #19: Protocol 6515

ihorhome1968, 2025-05-19:
Добрый день
Возможно ли добавить поддержку протокола 6515 ?
Инвертор Anenji 11kW 
Диагностический файл прикрепляю

[dess_monitor-01JVJ2TK60X3RN72CNGVQ0TN17-ИСело-d9759fe149f14e77d6cf9639d1c9eb97.json](https://github.com/user-attachments/files/20302750/dess_monitor-01JVJ2TK60X3RN72CNGVQ0TN17-.-d9759fe149f14e77d6cf9639d1c9eb97.json)
Сенсоры не подтягивает
[image]
Настройки подтягивает
[image]
Сырые данные - частично
[image]

С уважением...

Antoxa1081, 2025-05-19:
Приветствую, попробуйте установить последнюю версию 0.2.4 (сейчас выпустил в обновлении) и использовать режим Direct data request. Этот режим позволит быстрее обновлять данные, и получать более точные результаты. Что касается вашего devcode посмотрю дебаг, возьму в разработку на добавление. Спасибо за обратную связь.

ihorhome1968, 2025-05-20:
Спасибо за быстрый ответ и Вашу работу на благо всех! :)
Попробую, отпишу

ihorhome1968, 2025-05-20:
Не получается добавить интеграцию версии 0.2.4
Удалил DESS из HACS
Добавляю заново

[image]
Скачиваю, указываю версию 0.2.4

[image]
Добавляется

[image]
Перегружаю ХА полной перезагрузкой (пробовал и с перезагрузкой ХАОС)

[image]
Добавляю интеграцию, ввожу учетные данные и ставлю все птички
Показывает доступный инвертор, подтверждаю

Но добавляется версия 0.2.2 с сообщением об ошибке
"Не удалось настроить. Повторная попытка.: invalid literal for int() with base 16: 'null'"
[image]

ihorhome1968, 2025-05-20:
В журнале:
Эта ошибка возникла в пользовательской интеграции.

Регистратор: custom_components.dess_monitor.coordinators.direct_coordinator
Источник: helpers/update_coordinator.py:380
интеграция: DESS Monitor
Зафиксировано первый раз: 14:54:41 (18 случаев)
Зафиксировано последний раз: 15:14:16

Unexpected error fetching Direct request sensor data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/coordinators/direct_coordinator.py", line 119, in _async_update_data
    data_map = dict(await asyncio.gather(*map(fetch_device_data, self.devices)))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/coordinators/direct_coordinator.py", line 110, in fetch_device_data
    qpigs = await get_direct_data(token, secret, device, 'QPIGS')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/dess_monitor/api/helpers.py", line 557, in get_direct_data
    return decode_direct_response(cmd_name, result['dat'])
  File "/config/custom_components/dess_monitor/api/direct_commands.py", line 198, in decode_direct_response
    ascii_str = decode_ascii_response(hex_input)
  File "/config/custom_components/dess_monitor/api/direct_commands.py", line 6, in decode_ascii_response
    byte_values = bytes(int(b, 16) for b in hex_values)
  File "/config/custom_components/dess_monitor/api/direct_commands.py", line 6, in <genexpr>
    byte_values = bytes(int(b, 16) for b in hex_values)
                        ~~~^^^^^^^
ValueError: invalid literal for int() with base 16: 'null'

ihorhome1968, 2025-05-20:
Спасибо!

ihorhome1968, 2025-05-20:
Коллеги из соседних проектов пишут что подходит для ANJ-HHS-11 kW 48 V Wi-Fi
У меня именно такой... 
Но, не исключаю, что это не то что нам надо :)
[RS232.Communication.Protocol.V1.0-220209.pdf](https://github.com/user-attachments/files/20348817/RS232.Communication.Protocol.V1.0-220209.pdf)

Antoxa1081, 2025-05-20:
Благодарю, это то что нужно. Буду разбираться.

ihorhome1968, 2025-05-20:
Огромное Вам спасибо!

ihorhome1968, 2025-05-20:
Удялял - устанавливал заново, установил версию

[image]
и начали подтягиваться сенсоры

[image]
мистика :)

Antoxa1081, 2025-05-20:
Вероятно вы попали на "лаг" серверов десс монитора, они работают крайне нестабильно, из-за чего могла быть такая проблема. Временами я не мог добавлять свой аккаунт тоже, иногда помогает смена айпи адреса через впн. Через мобильный интернет чаще всего работает стабильнее чем от подводного

ihorhome1968, 2025-05-20:
Поковырялся...
В последней версии ошибка "Не удалось настроить. Повторная попытка.: invalid literal for int() with base 16: 'null'" проявляется если отмечена нижняя птичка в настройке интеграции:

[image]

Antoxa1081, 2025-05-21:
Выпустил новое обновление https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.2.6
В нем эта ошибка должна быть корректно исправлена для direct data request режиме. Если будет возможность попробуйте новую версию.

ihorhome1968, 2025-05-22:
Установил поверх, поставил птичку:

[image]

Удалил интеграцию, удалил в ХАКС
Добавил в ХАКС, перезапустил ХА, добавил интеграцию

[image]
Убрал птичку прямого запроса данных

[image]
сенсоры как и было - не все и РОУсенсоры тоже не все

[image]
То же самое было на предыдущей версии

Включил отладку, поставил птичку прямого запроса, ошибка повторилась, выключил птичку, ошибка пропала, выключил отладку, файл отладки прилагаю

[home-assistant_dess_monitor_2025-05-22T11-25-50.860Z.log](https://github.com/user-attachments/files/20391027/home-assistant_dess_monitor_2025-05-22T11-25-50.860Z.log)

Antoxa1081, 2025-05-22:
@ihorhome1968 

https://www.dessmonitor.com/#/generalNew/home
[image]

Если можете, попробуйте в дебаг интерфейсе отправить команду

`01 03 00 CA 00 03 25 F5`

Разбираюсь сейчас с документацией по протоколу документацию которого скинули

ihorhome1968, 2025-05-22:
Послал, в ответ получил
01 03 06 00 00 00 00 00 00 21 75

[image]

ihorhome1968, 2025-05-22:
Конечно

[image]

ihorhome1968, 2025-05-22:
[image]

Spick79, 2025-05-23:
Aceiași problemă cu invertor Easun smgii 11kw ce pot face?

ihorhome1968, 2025-05-23:
To hope and to wait :)

fiksx555, 2025-05-24:
https://github.com/robgt978/Easun-SMG-II-11Kw-esphome-/blob/main/esp32-easun-rs232.yaml

Antoxa1081, 2025-05-24:
@fiksx555 спасибо

fiksx555, 2025-05-24:
01 03 04 00 00 00 00 FA 33 01 03 04 00 00 00 09 3A 35 01 03 02 73 00 9D 74 01 03 02 00 03 F8 45 01 03 1A 39 32 42 33 32 35 30 31 31 30 30 34 36 36 00 00 00 00 00 00 00 00 00 00 12 51 9C C5 01 03 02 00 03 F8 45 01 03 02 00 00 B8 44 01 03 02 00 00 B8 44 01 03 02 13 89 74 D2 01 03 04 00 00 00 24 FA 28 01 03 02 13 89 74 D2 01 03 0A 02 2E 00 00 00 00 00 64 00 1C 10 CB 01 03 08 00 6F 00 00 00 00 00 1D CA D8 01 03 02 00 00 B8 44 01 03 02 00 00 B8 44 01 03 06 08 FB 00 24 00 1C 84 2B 01 03 10 08 FB 00 05 00 16 00 72 00 01 11 FF 00 00 00 00 65 A6 01 03 06 11 F3 00 02 00 7C 86 00 01 03 10 00 00 00 02 00 02 00 00 00 00 00 02 08 FC 13 88 5E 65 01 03 1E 00 02 02 76 00 03 00 00 00 00 00 00 00 03 02 2E 02 2E 00 00 04 B0 02 58 00 00 01 FE 01 F4 A5 47 01 03 06 01 E0 00 32 00 5F 40 85 01 03 0C 00 14 00 00 02 48 00 3C 00 78 00 1E CB 49 01 03 10 00 01 00 03 00 01 00 00 00 00 00 01 00 01 00 01 89 A6 01 03 04 00 00 00 00 FA 33 01 03 04 00 00 00 01 3B F3 01 03 12 07 E9 00 01 00 1D 00 04 00 39 00 0A 00 7A 00 00 06 02 EA 44

fiksx555, 2025-05-24:
это мониторинг посылает инвертору

fiksx555, 2025-05-24:
01 03 00 64 00 02 85 D4 01 03 00 6C 00 02 04 16 01 03 00 AB 00 01 F5 EA 01 03 00 B8 00 01 04 2F 01 03 00 BA 00 0D A5 EA 01 03 00 C9 00 01 54 34 01 03 00 CB 00 01 F5 F4 01 03 00 CE 00 01 E5 F5 01 03 00 E3 00 01 75 FC 01 03 00 E6 00 02 25 FC 01 03 00 FD 00 01 15 FA 01 03 01 15 00 05 95 F1 01 03 01 2E 00 04 25 FC 01 03 01 52 00 01 24 27 01 03 01 54 00 01 C4 26 01 03 01 56 00 03 E4 27 01 03 01 5A 00 08 65 E3 01 03 01 85 00 03 15 DE 01 03 02 58 00 08 C4 67 01 03 02 76 00 0F E5 AC 01 03 02 86 00 03 E5 9A 01 03 02 8A 00 06 E5 9A 01 03 02 A5 00 08 55 97 01 03 02 B1 00 02 95 94 01 03 02 B5 00 02 D4 55 01 03 02 B8 00 09 04 51

fiksx555, 2025-05-24:
[modbus_responses.xlsx](https://github.com/user-attachments/files/20423363/modbus_responses.xlsx)
Пропарсил даные. номера регистров и что в них находится. Anenji 11kWt

ihorhome1968, 2025-05-24:
В версии 0.2.8 существенно подтянулись сенсоры

[image]

Antoxa1081, 2025-06-01:
@ihorhome1968 здравствуйте, сможете попробовать отправить debug запрос который выше для пример давали

```
01 03 00 64 00 02 85 D4 01 03 00 6C 00 02 04 16 01 03 00 AB 00 01 F5 EA 01 03 00 B8 00 01 04 2F 01 03 00 BA 00 0D A5 EA 01 03 00 C9 00 01 54 34 01 03 00 CB 00 01 F5 F4 01 03 00 CE 00 01 E5 F5 01 03 00 E3 00 01 75 FC 01 03 00 E6 00 02 25 FC 01 03 00 FD 00 01 15 FA 01 03 01 15 00 05 95 F1 01 03 01 2E 00 04 25 FC 01 03 01 52 00 01 24 27 01 03 01 54 00 01 C4 26 01 03 01 56 00 03 E4 27 01 03 01 5A 00 08 65 E3 01 03 01 85 00 03 15 DE 01 03 02 58 00 08 C4 67 01 03 02 76 00 0F E5 AC 01 03 02 86 00 03 E5 9A 01 03 02 8A 00 06 E5 9A 01 03 02 A5 00 08 55 97 01 03 02 B1 00 02 95 94 01 03 02 B5 00 02 D4 55 01 03 02 B8 00 09 04 51
```

меня будет интересовать ответ который отдаст ваш инвертор на этот запрос, спасибо

fiksx555, 2025-06-01:
я же бросал ответы по этим запросам

fiksx555, 2025-06-01:
[Новый текстовый документ (5).txt](https://github.com/user-attachments/files/20541047/5.txt)
[modbus_responses3.xlsx](https://github.com/user-attachments/files/20541048/modbus_responses3.xlsx)

fiksx555, 2025-06-01:
там запросы и и ответы, в экселе розшифровки

fiksx555, 2025-06-01:
сори. это было адресовано не мне.

ihorhome1968, 2025-06-01:
Добрый день
Поздно увидел...
Достаточно того, что коллега выслал?
Или сделать тоже?

Antoxa1081, 2025-06-01:
@ihorhome1968 сделайте пожалуйста тоже, чтобы удостовериться что оно будет работать из интерфейса debug

filiphch, 2025-06-28:
I have the same protocol: a 11KW device and 6515 protocol
Direct communicqtion doesn't work
I'm using version 0.3.5
following this thread. will help where possible
thx for your huge work!