Antoxa1081/home-assistant-dess-monitor #22: devcode: 2376 Anenji 6.2kw Hybrid Inverter

barzick, 2025-05-22:
if i click on " Direct data request protocol (provide near-realtime direct data reading from equipment" no data can be charged with en error that i need to reload. If i only select "Device settings (allow to read & set equipment settings)" i have no error but no value is present.

Antoxa1081, 2025-05-22:
Can you provide diagnostic data file from device page on btn "Download diagnostics"
Perhaps there are interruptions on the server monitors now, so it is not possible to get all the data, or your inverter does not support the communication protocol, which is currently released in anticipation. In one of the manifestations of the threads I already had significant features of the Aneji inverters, this may also be a problem.

[image]

Antoxa1081, 2025-05-22:
@reallist93 добрый, вы дали немного не тот файл, не от интеграции dess_monitor

reallist93, 2025-05-22:
Почему то не качаются данные.

[17f08a47a6f8e241a7958d7cf4e2104b.txt](https://github.com/user-attachments/files/20389679/17f08a47a6f8e241a7958d7cf4e2104b.txt)

reallist93, 2025-05-22:
[config_entry-dess_monitor-01JQKTV7Y2EVT6G3MSQT43VKEZ.json](https://github.com/user-attachments/files/20389754/config_entry-dess_monitor-01JQKTV7Y2EVT6G3MSQT43VKEZ.json)

Вот вроде сачалось.

Antoxa1081, 2025-05-22:
@reallist93 

у вас это произошло после обновления? 
В старых версиях все работало нормально? 
С какой версии обновлялись?

reallist93, 2025-05-22:
Обновился, да сегодня.
Какая стояла ранее не могу сказать.

Antoxa1081, 2025-05-22:
@reallist93
а попробуйте на странице интеграции настроить

На странице интеграции "настроить", и выберите там галочкой ваш инвертор
[image]
[image]

reallist93, 2025-05-22:
Помогло..))

Теперь будет обновлять данные каждые 6 секунд?

[image]

barzick, 2025-05-22:
Actually i have the information back, but i neet unselect Direct data request. And i saw that the sources changed, i need to changed all my information in charts and script

[image]

Antoxa1081, 2025-05-22:
@reallist93 это косяк с моей стороны, я не учел миграцию со старых версий.. в будущем апдейте учту. Вы будете получать быстрые обновления для новых сенсоров с префиксом Direct * при включенной опции Direct Data request

Antoxa1081, 2025-05-22:
@barzick in new version names of old sensors not changed, only added a new sensors with prefix Direct * for direct mode. In future updated i think about merging fields, to reduce duplicated and availability direct data for energy meter sensors

reallist93, 2025-05-22:
> Direct * при включенной опции Прямой запрос данных
Если  устанавливаю галку на "Direct data request protocol (provide near-realtime direct data reading from equipment", то интеграция не запускается.

[image]

barzick, 2025-05-22:
ok sorry i doesn't saw that you have the same error, just saw the picture

Antoxa1081, 2026-10-30:
Direct data for SMG-II / Anenji / EASUN inverters (devcode 2376, Modbus RTU) is now supported. To use it, update to [v0.4.6](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.4.6), open **Settings → Devices & Services → DESS Monitor → Configure**, enable the direct data request option and set the direct protocol to **`smg2_modbus`** (the default `axpert` doesn't work with these inverters). The direct path still goes through the DESS cloud relay; for fully local polling see [home-assistant-dess-monitor-local](https://github.com/Antoxa1081/home-assistant-dess-monitor-local).

Closing for now. If direct sensors still don't update after that, please reopen and attach fresh diagnostics.

closed 2026-10-30 by Antoxa1081