Antoxa1081/home-assistant-dess-monitor #23: Пропала генерация

wsxshop24, 2025-05-22:
После последнего обновления пропала генерация солнечной энергии. Как это исправить?

[image]

Antoxa1081, 2025-05-22:
А что  показывает на графике PV Power?

wsxshop24, 2025-05-22:
Вот такой график 

[image]

Antoxa1081, 2025-05-23:
На графике сенсора PV Power который в интеграции  home assistant

wsxshop24, 2025-05-23:
[image]

wsxshop24, 2025-05-23:
У меня сейчас вот такие настройки

[image]

Antoxa1081, 2025-05-23:
Проверьте сенсор PV In Energy а так же настройт панели энергии, возможно у сенсора имя поменялось

wsxshop24, 2025-05-23:
[image]

wsxshop24, 2025-05-23:
Подскажите, пожалуйста, где конкретно посмотреть эти настройки панели энергии

Antoxa1081, 2025-05-23:
PV Total Energy предоставляется с облака dessmonitor, не рекомендую его использовать так как у них часто такие приколы бывают с подсчётами. Лучше использовать PV In Energy, этот сенсор считает энергию по графику PV Power

Antoxa1081, 2025-05-23:
[image]
[image]

wsxshop24, 2025-05-23:
Заменил настройку. Какая-то генерация появилась. Нужно понаблюдать

[image]

Antoxa1081, 2025-05-23:
@wsxshop24 нашел со своей стороны мисс по апдейту девайсов, в 0.2.7 апдейте поправил, но рекомендую использовать сенсор PV In Energy, он работает точнее, и не подвержен ошибкам со стороны облака dess (у них на облаке вначале весны неправильно считалась генерация)

wsxshop24, 2025-05-23:
Вроде как, все работает) Спасибо большое!

closed 2025-05-23 by wsxshop24