Antoxa1081/home-assistant-dess-monitor #24: devcode: 2376 Direct data request protocol for Anenji 6.2kw Hybrid Inverter

reallist93, 2025-05-23:
Добрый день.
В очередной раз благодарю за Ваш труд.
Подскажите плз. Direct data request protocol заработает на Anenji 6.2kw ?
Или уже и ждать не стоит?
Заранее спасибо.

Antoxa1081, 2025-05-23:
Добрый, спасибо за ваш отзыв. В данный момент работаю над этим, возможно на следующей неделе получится с этим что-то, но не обещаю. У Anenji modbus протокол в отличии от вольтроник-подобных (Axpert подобных ASCII), пока не понятно как wifi datalogger debug пересылает ему сообщения.

reallist93, 2025-05-23:
Спасибо.. 
Буду ждать)))

reallist93, 2025-05-23:
Да , всё работает кроме указанного выше.
Настройки интеграции необходимо установить как на скрине ниже .

[image]

[image]

foziewhite, 2025-05-23:
> Да , всё работает кроме указанного выше. Настройки интеграции необходимо установить как на скрине ниже .
Спасибо, вы меня очень выручили

closed 2025-05-26 by reallist93