Antoxa1081/home-assistant-dess-monitor #28: Dessmonitor link changed?

xoomas, 2025-05-30:
Failed setup, will retry: 404, message='Attempt to decode JSON with unexpected mimetype: text/html', url='https://web.dessmonitor.com/remote/?sign=1298e32a49205463677a886f5a45303cfd74.. (didn't copy all link) just before update it was not working :) so thought maybe that's why new update came out but still same. Any idea?

Luukullus, 2025-09-13:
Everything fine for me.
Just set it up and everything works fine (PowMr 10.2kw All in One Solar Inverter)

Alf-Tupper, 2025-09-25:
Ive started getting this error Failed setup, will retry: 502, message='Attempt to decode JSON with unexpected mimetype: text/html', url='https://web.shinemonitor.com/public/?

Haven't been able to reconnect, wouldn't accept my login details at first, but after it did, I got the error above.

Antoxa1081, 2025-09-26:
> Ive started getting this error Failed setup, will retry: 502, message='Attempt to decode JSON with unexpected mimetype: text/html', url='https://web.shinemonitor.com/public/?
> 
> Haven't been able to reconnect, wouldn't accept my login details at first, but after it did, I got the error above.

https://github.com/Antoxa1081/home-assistant-dess-monitor/issues/66#issuecomment-3337321581