Antoxa1081/home-assistant-dess-monitor #32: DevCode: 2452 Direct Data Support.

umarjamilpc, 2025-06-05:
Hello, First of all i thank you for supporting this project it help me alot to integrate this into my home assistant. Is there any way to get direct data from my inverter ?. Currently from dess-monitor i am getting data with 5 mins delay. is it possible to get real time data.

Inverter Site Link: https://maxpower.com.pk/product/suntronic-pv-7000-pro

[image]

umarjamilpc, 2025-06-11:
Hello, i have check direct data request but it shows unknown values. Can you help me how i can check protocols and QPIGS ?
do i need to use putty to check it or is there any other way ?

umarjamilpc, 2025-06-27:
Found out that my inverter is using Protocol 18, also tell me what else you need ?

Antoxa1081, 2026-05-03:
Changed to https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.4.3

Antoxa1081, 2026-10-30:
PI18 direct data for devcode 2452 has been supported since v0.4.3 (confirmed working above). For truly local polling without the cloud, see [home-assistant-dess-monitor-local](https://github.com/Antoxa1081/home-assistant-dess-monitor-local). If specific PI18 sensors are still unknown on [v0.4.6](https://github.com/Antoxa1081/home-assistant-dess-monitor/releases/tag/v0.4.6), please open a new issue with diagnostics.

closed 2026-10-30 by Antoxa1081