Antoxa1081/home-assistant-dess-monitor #36: pv2 unidenty fied data show

ajaydabas, 2025-06-13:
In the 3.0 kW inverter, there is only PV1, but PV2 is providing data on its own, which is not present in the sensor; there is only 1 MPPT.
2477 dev code[

[dess_monitor-01JWECJQBW65YCRX14J3NNV40T-Home Inverter 3.0kW-9f735bac88f9bc7424e28cd49b2981c9.json](https://github.com/user-attachments/files/20729978/dess_monitor-01JWECJQBW65YCRX14J3NNV40T-Home.Inverter.3.0kW-9f735bac88f9bc7424e28cd49b2981c9.json)

]

Antoxa1081, 2025-06-19:
Hello, please tell me your inverter model, to clarify its decoding of registers/direct request fields

ballhtml, 2026-11-03:
Hi @ajaydabas and @Antoxa1081,

This occurs because 3.0 kW inverters (such as Muscle Grid / Axpert VM 24V on **devcode 2477**) are hardware-designed with **single-MPPT only (PV1)**, but the integration's default schema discovers both PV1 and PV2 channels by default.

### Solutions:

1. **For users right now (Home Assistant UI):**
   * Go to **Settings -> Devices & Services -> Entities**.
   * Search for `PV2`.
   * Click on the `PV2 Voltage` and `PV2 Power` entities and select **Disable Entity**. Home Assistant will remove them from your dashboards and energy cards immediately.

2. **For @Antoxa1081 in `mapping/seed.py` / device definitions:**
   * Devcode `2477` should be explicitly tagged as `single_mppt: True` so that the entity loader skips generating `pv2_voltage`, `pv2_current`, and `pv2_power` for this family.