Antoxa1081/home-assistant-dess-monitor #37: login issue

cuietviper, 2025-06-13:
When I try to log in with my DESS credentials, I get the following error. Since I am new to this, could you please help

[image]

justindeweger, 2025-06-19:
I Have the same. But it is weird, I use a QR Code to logon at dessmonitor.com (from the smartESS app). And use a username and password in de ios app SMARTESS. But if I logon with those credentials on DESSmonitor.com the username is incorrect. I can logon there with an old password but then I don't see my hybrid inverter.

Logon in this integration is not working with the old password (i think he don t see an inverter).
writing the new password from the smartESS app is not working in this integration.

Any suggestions to get ride of 2 accounts?

Antoxa1081, 2025-06-19:
Unfortunately I can't help you with the problem of connecting accounts. For the integration to work correctly, you need the account to log in normally with the login and password through the web version of Dess.

cuietviper, 2025-06-22:
Well in my case. I have logged in with the username and password. But still this issue persists

viper71285, 2025-06-23:
Well, my case is on this is differnt as well.
I can log in on SMart ESS, but on the website I get a error "The system is abnormal. Please contact the administrator" I have never been able to get logged in on dessmonitor.com but the intergration works partially. Just not showing all the values that are available.

justindeweger, 2025-06-25:
I deleted in both accounts (username pass and the qr) my devices.

Added the device by laptop by www portal.
 
Waited for 30 min. And it showed up. After that I could logon at home assistant and this integrations

Now it is working!

Really cool, now I can set up automations and notifications!

Thanks! Awesome!

closed 2025-08-07 by Antoxa1081