Antoxa1081/home-assistant-dess-monitor #46: PV2 data n/a devcode 6416

mclaudod, 2025-06-26:
Добрый.

freeewilly, 2026-04-18:
I just updated to 0.3.4.7, but the issue with "unknown" PV2 Voltage and PV2 Power unfortunately remains. Devcode 6416.
Thanks a lot in advance!

ballhtml, 2026-10-31:
Hi @Antoxa1081, @vtorresursy5-a11y, and @ctyber82,

Following up on the helpful findings from @vtorresursy5-a11y and @ctyber82 above regarding devcode 6416:

Checking the current `main` branch (`mapping/seed.py`), `pv2_power` already has the candidate entry for `eybond_read_2` nicely in place:
```python
"pv2_power": [
    ...
    # MiC 6.2 (devcode 6416) — PV2 input power lives under a generic "eybond_read_2" alias
    ProviderKeyCandidate("eybond_read_2", MATCH_FIELD_ID),
    ProviderKeyCandidate("eybond_read_2", MATCH_FIELD_PAR),
]
```

To complete the companion voltage leg for this firmware family, we can add `eybond_read_1` under `"pv2_voltage"`:

```diff
     "pv2_voltage": [
         ProviderKeyCandidate("bt_voltage_2", MATCH_FIELD_ID),
         ProviderKeyCandidate("bt_voltage_2", MATCH_FIELD_PAR),
         ProviderKeyCandidate("PV2 Input voltage", MATCH_FIELD_ID),
         ProviderKeyCandidate("PV2 Voltage", MATCH_FIELD_ID),
         ProviderKeyCandidate("PV2 Input Voltage", MATCH_FIELD_ID),
         ProviderKeyCandidate("PV2 Input Voltage", MATCH_FIELD_PAR),
         # Anenji 11kw (devcode 6544).
         ProviderKeyCandidate("eybond_read_43973", MATCH_FIELD_ID),
         ProviderKeyCandidate("eybond_read_43973", MATCH_FIELD_PAR),
         # Anenji HHS 11kW (devcode 6515) — PV2 Voltage, pars block only.
         ProviderKeyCandidate("eybond_read_46", MATCH_FIELD_PAR, unit_hint="V"),
+        # MiC 6.2 / Anern EVO 10.2 (devcode 6416) — PV2 Input Voltage.
+        ProviderKeyCandidate("eybond_read_1", MATCH_FIELD_ID),
+        ProviderKeyCandidate("eybond_read_1", MATCH_FIELD_PAR, unit_hint="V"),
         # Stevo Hybrid Monster (devcode 6467) — counterintuitive: id
         # `pv_voltage_1` is the PV2 string.
         ProviderKeyCandidate("pv_voltage_1", MATCH_FIELD_ID),
     ],
```

This pairs naturally with the existing `eybond_read_2` power mapping and should allow the integration to resolve PV2 Voltage for devcode 6416 seamlessly!

ivdome, 2026-11-01:
Same problem. No PV2 voltage, power and current...