Antoxa1081/home-assistant-dess-monitor #47: EASun inverter SMH-III 6.2kW devcode: 6416

fcguilherme-star, 2025-06-29:
The integration is failing, and giving the following error:

2025-06-29 02:12:52.997 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry xpto for dess_monitor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/dess_monitor/__init__.py", line 24, in async_setup_entry
my_coordinator = MainCoordinator(hass, entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/dess_monitor/coordinators/coordinator.py", line 30, in __init__
super().__init__(
TypeError: DataUpdateCoordinator.__init__() got an unexpected keyword argument 'config_entry'

Antoxa1081, 2025-06-29:
Hello, please attach the diagnostic configuration files (download diagnostics button on the device and integration page)

fcguilherme-star, 2025-06-30:
Hello, here is the diagnostics file:

[config_entry-hacs-01JYWH9HTTPY7PK4B6QR2QXHED.json](https://github.com/user-attachments/files/20970715/config_entry-hacs-01JYWH9HTTPY7PK4B6QR2QXHED.json)

adargains, 2025-09-09:
Hi,

I have the same need for devcode 6416. However, my inverter is a SUMRY ECO-LV 3.6KW PLUS. I am away from the inverter that is off now. I will post my diag configuraltion file this saturday.

adargains, 2025-09-16:
> [@adargains](https://github.com/adargains) [@fcguilherme-star](https://github.com/fcguilherme-star) Make sure you are actually running Home Assistant 2024.11 or newer

My HA is fully updated. Please, Check log and diag files attached:

[home-assistant_dess_monitor_2025-09-14T02-42-09.985Z.log](https://github.com/user-attachments/files/22370388/home-assistant_dess_monitor_2025-09-14T02-42-09.985Z.log)
[dess_monitor-01K40Q2Y4CSW3PG1CM4TH8QXET-SUMRY ECO-LV 3.6KW PLUS-e84afcfb2f89dca59e2b0cafa4ac2b3b.json](https://github.com/user-attachments/files/22370387/dess_monitor-01K40Q2Y4CSW3PG1CM4TH8QXET-SUMRY.ECO-LV.3.6KW.PLUS-e84afcfb2f89dca59e2b0cafa4ac2b3b.json)