Antoxa1081/home-assistant-dess-monitor #52: Cannot Configure devices

AURELIODINVERNO, 2025-07-14:
Hello I have 3 PowMR 6.2 inverter running in parallel each with one wifi sensor (showing all data in SmartESS).

Tried to integrate them in HA, followed the procedure till the configuration step, which ended in an error.

here below the log. Any idea what could have gone wrong?

Thanks in advance for your reply :-)

Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 5:31:33 PM (5 occurrences)
Last logged: 6:00:36 PM

Error setting up entry aur*****no@gmail.com for dess_monitor
Traceback (most recent call last):
  File "/snap/home-assistant-snap/637/lib/python3.12/site-packages/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/snap/home-assistant-snap/637/custom_components/dess_monitor/__init__.py", line 24, in async_setup_entry
    my_coordinator = MainCoordinator(hass, entry)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/snap/home-assistant-snap/637/custom_components/dess_monitor/coordinators/coordinator.py", line 30, in __init__
    super().__init__(
TypeError: DataUpdateCoordinator.__init__() got an unexpected keyword argument 'config_entry'

Antoxa1081, 2025-07-14:
Hi, what version of home assistant?

AURELIODINVERNO, 2025-07-14:
came with latest LTM ubuntu (running in ubuntu with Firefox) : version 2024.10.4